mrjohnk/PMW3360DM-T2QU #1: Problem with teensy

robotpapier, 2017-07-06:
Hi,
I bought the sensor on tindie and
I succeeded to get the sensor to work with arduino uno but couldn't make it work with teensy 3.2.
The led doesn't even lit.

The serial monitor shows :
Uploading firmware...
Optical Chip Initialized
---
Product_ID
0
11111111
FF
---
Inverse_Product_ID
3F
11111111
FF
---
SROM_Version
2A
11111111
FF
---
Motion
2
11111111
FF

and movement is not updated.

Can someone please help me with that?

Thanks

mrjohnk, 2017-07-06:
It looks like the wiring or settings are not quite right since the output
is indicating that there is no valid SPI communication taking place.  I too
use the Teensy to test all boards with, so it will work once the settings
and wiring are right.

On the Teensy 3.2, these pins should be wired

DIN to MISO (12)
DOUT to MOSI (11)
CS to SS (10)
SCK to SCK (13)
0 to MOT (0)
3.3V to VIN
GND to GND

gstoehls, 2017-10-11:
Dear mrjohnk,
I followed instructions and wired as you say.
I receive this output on the arduino serial monitor:

Uploading firmware...
Optical Chip Initialized
---
Product_ID
0
10100010
A2
---
Inverse_Product_ID
3F
10100010
A2
---
SROM_Version
2A
10100010
A2
---
Motion
2
10000000
80

No update on movements either...

Can you help?

Regards
gstoehls

gagankhandate, 2018-03-29:
Hi gstoehls,

Did you get this working ? I am currently facing exactly the same issue. Would be great if you could share your solution.

mrjohnk, 2018-03-29:
Can you provide details about what platform you are using and what pins you
connected on both sides?

gstoehls, 2018-04-03:
Hi gagankhandate,

Can you share your working code?

Regards
gstoehls

mrjohnk, 2018-04-03:
The code that is shared is compatible with Teensy.  You probably need to
set the two options in the code to properly indicate which pins on your
Teesy your sensor is connected to.  These are:

//Set this to what pin your "INT0" hardware interrupt feature is on
#define Motion_Interrupt_Pin 3     // (MOT pin)

//This is the SPI "slave select" pin that the sensor is hooked up to
const int ncs = 10;      // (SS pin)

If you continue to have troubles, please indicate which Teensy you are
using and what pins are connected between the sensor and Teensy

aacemoglu, 2018-10-03:
Dear mrjohnk, 
I have the same problem with Teensy 3.5. 
[image]
I connected the sensor to teensy 3.5 like this. (please see attached png)

VI = 3.3V                       
MI   = MISO0 -or- DIN      | PIN   12
MO   = MOSI0 -or- DOUT | PIN   11
SC = SCK0 -or- SCK           | PIN   13
SS = CS0 -or- CS               | PIN 15
MT = Pin 0                        | PIN 0
GD = GND 

And I modified the code like this;
//Set this to what pin your "INT0" hardware interrupt feature is on
#define Motion_Interrupt_Pin 0
const int ncs = 15;  //This is the SPI "slave select" pin that the sensor is hooked up to

When I upload the skecth, I get only this message, no data from sensor.

Optical Chip Initialized
---
Product_ID
0
10100001
A1
---
Inverse_Product_ID
3F
10100001
A1
---
SROM_Version
2A
10100001
A1
---
Motion
2
10000000
80

I would be really happy, if you can help how I can solve this issue. 

Thanks 
Alp