mrjohnk/PMW3360DM-T2QU #2: Changeable DPI

CanuckianGamer, 2017-09-21:
Can we change the dpi of the sensor?

mrjohnk, 2017-09-21:
The sensor has configurable registers that control the CPI (commonly also
called DPI as well).  The datasheet details how to do this.

CanuckianGamer, 2017-09-21:
What would that be under?

mrjohnk, 2017-09-21:
This is handled in configuration register "Config1" (0x0F).  It increments
at 100 CPI at a time between 100 CPI and 12000 CPI.  Please see page 38 of
the datasheet.

CanuckianGamer, 2017-09-21:
So you change the value at the top where it has #define?

mrjohnk, 2017-09-21:
Actually no, those are the friendly name assignments for the number of each
register in the chip.  You would need to calculate a new hex value and
preform an SPI write to this register to make a change in the CPI.

CanuckianGamer, 2017-09-22:
I'm confused

mrjohnk, 2017-09-22:
Here is an example of how to set the CPI:
  adns_write_reg(Config1, 0x15);

In place of the 15 in the example, valid values are anything between 0 - 77
in hexidecimal (base 16).

CanuckianGamer, 2017-09-22:
I can use that as many times as i want right like for a dpi button?

mrjohnk, 2017-09-22:
Initially, you can put that at the bottom of the startup section to see how
it works.  Overall though, yes, you can use this as much as you want to
change the CPI in real-time, anytime after the chip is initialized.  Put
some logic behind a button to increase or decrease the CPI as needed.

closed 2017-09-24 by CanuckianGamer