mrjohnk/PMW3360DM-T2QU #4: update on movements

gstoehls, 2017-10-25:
Dear mrjohnk,
I followed instructions and wired the pins on my teensy 3.2 as you say.
I receive this output on the arduino serial monitor:

Uploading firmware...
Optical Chip Initialized

Product_ID
0
11010111
D7

Inverse_Product_ID
3F
11010111
D7

SROM_Version
2A
11010111
D7

Motion
2
10000000
80

But no update on movements either...

Can you help?

Regards
gstoehls

mrjohnk, 2017-10-26:
Can you tell me the actual pin numbers on the Teensy 3.2 that you used per
the diagram for the 3.2 version on this page:
https://www.pjrc.com/teensy/pinout.html

gstoehls, 2017-10-26:
Thank you for your answer.
Here's what I did:

PMW3360 --- Teensy 3.2:

RS --- not connected
GD --- GND
MT --- PIN(0)
SS --- CS(10)
SC --- SCK(13)
MO --- DOUT(11)
MI --- DIN(12)
VI --- 3.3V

Here's a picture to illustrate the pin assignment:
[image]

mrjohnk, 2017-10-26:
I went over that a few times this morning, but couldn't find anything you
had hooked up incorrectly.  Regarding the software, did you use it without
modification and did you add the SROM file as a tab inside the main sketch?

gstoehls, 2017-10-27:
Yes, the ROM was added as a tab. The software was not modified, except for the MT pin definition. I defined it as pin 0 instead of pin 3 (as in the original file) but that didnt help, unfortunately.

jshay21, 2018-12-16:
@mrjohnk I am facing issues also trying to get x and y values.  I am using Teensy 3.2, have the pins set correctly (checked 5 times), set MT to pin 0 in hardware and in code, can see red IR light, have rang and verified cables on each pin to make sure that wasn't an issue, and have tried interrupt and polling version of the example.  You can see my result here with the polling version.  When i move my finger over the sensor I can get some data from the y axis, and x axis does not change at all.

https://www.dropbox.com/s/5ngwazaqn4dze6q/2018-12-16_14-47-26.mp4?dl=0

Is there anything you can think of that might be wrong?   I am running arduino 1.8.7 IDE

raycewest, 2019-01-19:
I just received a Teensy 3.2 and can also confirm the exact same hookup and firmware (only changing the MOT PIn to 0 in the code) produces the same issue - initialization output looks good, but no movement reported.

raycewest, 2019-01-19:
Can also confirm that I see the -128's on the serial monitor, which can't be right.  Did you ever get it working on the Teensy?

levihilton, 2019-03-08:
I seem to be having a similar issue.  I am working with a Teensy 3.6.  The monitor shows this a the beginning
```
Optical Chip Initialized
---
Product_ID
0
10000000
80
---
Inverse_Product_ID
3F
10000000
80
---
SROM_Version
2A
10000000
80
---
Motion
2
10000000
80
```

then continues to show 
`x = -128 | y = -128`
over and over again.

My pin orientation is 
```
VI = 3.3V between pins 12 and 24
MI = MISO0 pin 12
MO = MOSI0 pin 11
SC = SCK0 pin 13
SS = CS0 pin 10
MT = Pin 0
GD = GND
```
There is no light on on the sensor that I can see either.  Any suggestions?

mrjohnk, 2019-03-08:
Try to slow down the SPI clock.  I recently had some luck with this on a
Teensy LC.  Here is what the SPI init would look like for that:

SPI.begin();
SPI.setDataMode(SPI_MODE3);
SPI.setBitOrder(MSBFIRST);
SPI.setClockDivider(SPI_CLOCK_DIV128);
//SPI.setClockDivider(4);

levihilton, 2019-03-08:
That is what it is currently set at.  Would increasing any of the delays in perform startup make a difference?

mrjohnk, 2019-03-08:
I doubt changing those will help.  You may have a wiring issue since all of
the data retrieved is the same in the various start-up banner
troubleshooting items.  I don't see evidence of a working SPI bus.  Please
email me details about what MCU you are using and how you have wired up the
connections between it and the sensor. Additionally, what have you set the
ncs variable to, which is the slave select pin.  Send a picture if you want
as well.  Support@MrJohnk-dot-Com

levihilton, 2019-03-08:
I believe I have found a solution.  Rather than using the following:
```
SPI.setDataMode(SPI_MODE3);
SPI.setBitOrder(MSBFIRST);
SPI.setClockDivider(SPI_CLOCK_DIV128);
```
Use something like:
`SPI.beginTransaction(SPISettings(2000000, MSBFIRST, SPI_MODE3));`
I believe this slows down the clock to 2MHz, which is the max serial port clock frequency for this particular sensor according to the datasheets.  I have yet to do extensive testing, but this may be a solution for those having difficulties.

raycewest, 2019-03-12:
@levihilton that seems to help with the teensy.  Could you confirm or deny the behavior I describe in https://github.com/mrjohnk/PMW3360DM-T2QU/issues/22?

raycewest, 2019-03-12:
Thanks for replying btw.