mrjohnk/PMW3360DM-T2QU #7: Schematic in PMW3360DM-1.3.pdf wrong?

ErwinRieger, 2018-01-12:
Hi, 

the updated schematic in the file [PMW3360DM-1.3.pdf](https://github.com/mrjohnk/PMW3360DM-T2QU/blob/9fc3bb78332c322a4703ac363594b76deda3ea62/PMW3360DM-1.3.pdf) seems to be wrong.

Connecting SS-H to the DIR pin of the MISO transceiver should be enough to share the SPI bus with other devices.

But you also swapped the power lines (VCCA <-> VCCB) and the A/B lines (compared to the original cirquit in PMW3360DM.pdf). This is not neccessary and will not work.

mrjohnk, 2018-01-12:
You are correct.  I made a follow up version 1.4 that corrected this.
Essentially, the idea is to disconnect the MISO line when the module is not
selected as it tends to hold onto the bus even when it isn't selected and
causes trouble for other devices trying to share the same bus.  In version
1.3, I had the polarity of the SS line wrong as it relates to the switching
of the level shifter.

ItsssJustice, 2018-01-24:
Could you please upload your version 1.4 schematic? I'm assuming that the version is still up is indeed the incorrect 1.3. Thanks in advance!

mrjohnk, 2018-01-25:
I've added v1.4 to the repository

ErwinRieger, 2018-01-25:
An fixed schematic is available, so closing this.

closed 2018-01-25 by ErwinRieger