mrjohnk/PMW3360DM-T2QU #8: adns_upload_firmware(): How to verify SROM_ID?

ErwinRieger, 2018-01-12:
adns_upload_firmware(): After uploading the firmware blob, the SROM_ID is read back but it is not verified that it is the expected value.

Which firmware ID should be reported for the firmware in SROM_0x04_Arduino.ino? Should it be 0x04?

I tried to compare the returned SROM_ID with 0x04, but this isn't working. The returned ID is 0x68. Firmare upload seems ok because the sensor is working and is reporting reasonable values.

Can you tell us the expected firmware ID, or even better: update (and test) the arduino code with the SROM_ID check.

kabrio, 2018-02-13:
Have you seen this?

[PMW3360DM-T2QU/proper startup banner.txt](PMW3360DM-T2QU/proper startup banner.txt)

mrjohnk, 2018-02-14:
The SROM version should report "4"

ErwinRieger, 2018-02-14:
OK, i meant something else:
if you modifiy _adns_upload_firmware()_ in the Arduino example to output the SROM_ID that is read back just after uploading the firmware, like this:

```
...
168c167,168
<   adns_read_reg(SROM_ID);
---
>   Serial.print("Sromid in adns_upload_firmware(): ");
>   Serial.println(adns_read_reg(SROM_ID));
...

```
Then you get this output (note the unexpected srom id of 255 and later on srom id 4):

```
Uploading firmware...
Sromid in adns_upload_firmware(): 255
Optical Chip Initialized
...---
SROM_Version
2A
100
4
---

```

ErwinRieger, 2018-02-14:
This pullrequest should fix it: https://github.com/mrjohnk/PMW3360DM-T2QU/pull/13