mrjohnk/PMW3360DM-T2QU #12: Using INT1 instead of INT0

kabrio, 2018-02-13:
```
//Set this to what pin your "INT0" hardware interrupt feature is on
#define Motion_Interrupt_Pin 1
```
I am trying to use the senor with an Teensy 2.0++
I was wondering if it is also possible to use INT1 instead of INT0.

kabrio, 2018-02-13:
After some more tests: I guess not.
Interesting (to me at least) even if I set Motion_Interrupt_Pin to INT1 in code, I only get values from sensor when I connect MT to INT0.

mrjohnk, 2018-02-14:
I have used different interrupts on the Teensy 2.0 before with no problem.
You just need to hook to the correct related pin and indicate which
interrupt to watch for on the attachInterrupt line in the code.  Also don't
forget to raise the pull-up on that pin in input mode.

kabrio, 2018-02-14:
Thanks for the answer!
Looked into the code again and I think the problem is on [line 76](https://github.com/mrjohnk/PMW3360DM-T2QU/blob/ca68ad735092b23a22075d60d1da6456a94e183a/Arduino%20Examples/PMW3360DM-Interrupt/PMW3360DM-Interrupt.ino#L76).
There is 0 hardcoded where probably should be _Motion_Interrupt_Pin_ used instead.
Now works with other interrupts than 0 too!