mrjohnk/PMW3360DM-T2QU #14: Firmware not uploaded to Teensy++ 2.0 ?

kabrio, 2018-02-14:
I am trying to use the sensor board form Tindie with a Teensy++ 2.0.

The info I get from serial makes me wonder, it does not correlate with the contents of "proper startup banner.txt" so I wonder if the firmware gets uploaded at all?

Here's how I connected the board to Teensy++ 2.0:

VI  -  5V
MI  -  B3/23/MISO 
MO - B2/22/MOSI 
SC - B1/21/SCLK
SS - B0/20/SS
MT - D0/0 (I set Motion_Interrupt_Pin in code to 0)
GD - GND
RS - not connected

Here's what I get in from serial:
Uploading firmware...
Optical Chip Initialized
---
Product_ID
0
11111111
FF
---
Inverse_Product_ID
3F
10111101
BD
---
SROM_Version
2A
0
0
---
Motion
2
0
0

**Does this mean, that the firmware does not get uploaded?**

However,  I am able to get some values from the sensor, but the mouse movements are not very satisfying. Might be a problem with how I installed the sensor though, not sure.

Thanks for any help!

mrjohnk, 2018-02-15:
Your wiring looks right.  I suspect that you need to change the "ncs"
variable to 20 since that is where you have the SS pin wired to

kabrio, 2018-02-15:
Thanks for the answer!

Sorry, I forgot to mention that I set ncs variable to 20. 

Is it normal that the firmware upload fails but the sensor still kinda works?

mrjohnk, 2018-02-16:
Generally, the sensor either works or not.  The lens needs to be attached,
the distance to the surface needs to be exactly 2.4mm and the surface
itself has to be of the proper quality.  Generally, it needs to not be
shiny and have enough texture that can be tracked by the sensor.

kabrio, 2018-02-16:
So i just need to worry about the distance to surface and the right surface even if the output from serial doesn't match the contents of "proper startup banner.txt" ?

mrjohnk, 2018-02-16:
The startup banner I posted is just what I get.  What is different about
the one you get? It may be OK, but it would be a good idea to decode the
differences using the datasheet to see if there is a problem or not

kabrio, 2018-02-21:
> The lens needs to be attached,
> the distance to the surface needs to be exactly 2.4mm and the surface
> itself has to be of the proper quality.

Okay, just realized, that I will need something like a base plate to get the right distance. Always thought the lens takes care of proper distance...