mrjohnk/PMW3360DM-T2QU #15: Base plate 3D drawing for printing

kabrio, 2018-03-08:
Has anyone here already designed a base plate for the sensor module for printing on a 3d printer?
Would be rad if it could be shared (maybe on thingiverse?)

Also interested in any experience with making a base plate. How did you make it work? 
Where there any problems?

Thank you!

ItsssJustice, 2018-03-08:
I am currently working on a design for this as I speak, I'll get back to you once I've finished the design and tested it out if this thread is left open!

kabrio, 2018-03-08:
that'd be great! maybe should ask around at some diy electronics forums too..

ItsssJustice, 2018-03-12:
Here is a SolidWorks file in the attached zip folder (with related STL if you simply want to just get it printed). I have shown an image of my version of this board (that this stand-off was made for), keep in mind that the screw holes I added will likely need adjusting for your personal circuitboard. Results may vary from printer to printer as the tolerance size of the printer I have access to is rather poor, so this won't be the holy grail solution - it will require tweeking based on the printer you use. I don't want to post any information other than images showing the design of the stand-off relating to my own version of the board - as mrjohnk is selling boards from his designs and is kindly providing this github as open access.

[image]

[image]

[image]

[PMW3360-T2QU CAD Stand-Off Plate.zip](https://github.com/mrjohnk/PMW3360DM-T2QU/files/1804714/PMW3360-T2QU.CAD.Stand-Off.Plate.zip)

SunjunKim, 2019-03-29:
I made a 3D printable baseplate for the recent board design.
Check this:
https://cad.onshape.com/documents/be34d386d9b3899b60f7ef3d/w/8987d358111277c2f626d8dd/e/c91ca5bfdd280a5bc30b85f1