mrjohnk/PMW3360DM-T2QU #17: Issue Using with Arduino Zero

redknight1056, 2018-04-28:
Hello.
I've also bought your sensor on tindie and in a similar way to the first 'issues' post I succeeded to get the sensor to work with Arduino Uno but couldn't make it work with Arduino Zero.

I get the following output:
`Uploading firmware...
Optical Chip Initialized
---
Product_ID
0
11111111
FF
---
Inverse_Product_ID
3F
11111111
FF
---
SROM_Version
2A
11111111
FF
---
Motion
2
11111111
FF
`
I'm using the following pinouts:
DIN to MISO (12) 
DOUT to MOSI (11) 
CS to SS (10) 
SCK to SCK (13)  
MOT to pin(3)   
VIN GND to GND

mrjohnk, 2018-04-28:
Could be that you need to connect the sensor to another place on the board
to access the SPI bus?  Please see this image:
https://learn.adafruit.com/assets/29968

redknight1056, 2018-04-28:
Ahh great find.

That alone was 50% the issue solved.

Turns out you have to use the following to achieve a working sensor:
SPI.beginTransaction(settingsA);
SPI.endTransaction();
// set up the speed, data order and data mode
SPISettings settingsA(2000000, MSBFIRST, SPI_MODE3);

redknight1056, 2018-04-28:
My only issue is around the last the following:
//Motion
//2
//100
//4

There is only one issue around LED not lighting and the motion

redknight1056, 2018-04-28:
The rest of the output around 'Product_ID' etc is the same as your example output file.

redknight1056, 2018-04-29:
Moving this along I tried using the sensor with a Sparkfun Pro Micro 3.3V and I get the correct output. But the movement output is like 0 or 2 X and 0 or 2 Y. Is there something I'm missing?

redknight1056, 2018-04-30:
Hmm very weird as the polling example works lovely.  Apart from using the poll timer to read the xydat is there any other differences.

mrjohnk, 2018-05-01:
The polling really just checks to see if there is any new movement data
available on a regular frequency and does not rely upon a hardware
interrupt to trigger the MCU read the motion data registers.  As such, the
MOT pin is not used on the polling example.  So, it is a bit less
effecient, but that is about the only difference.