mrjohnk/PMW3360DM-T2QU #21: Teensy 3.5 - what should be the clock rate?

aacemoglu, 2018-10-15:
Dear mrjohnk, 
I tried this code with Arduino Uno, it works without any problem. 
But I have some problems when I use Teensy 3.5. 

I connected the pins like this:
/*
 *                                          - Teensy 3.5
 * VI = 3.3V                          - 3.3V
 * MI = MISO0 -or- DIN       - 12
 * MO = MOSI0 -or- DOUT  - 11
 * SC = SCK0 -or- SCK          - 13
 * SS = CS0 -or- CS               - 10
 * MT = Pin 0                        - 0
 * GD = GND                        - GND
 */

Then, I modified the code to SPI.beginTransaction() and SPI.endTransaction() : 

--------------------------------------------------------------
SPISettings settingsA(2000000, MSBFIRST, SPI_MODE3);

void adns_com_begin(){
  SPI.beginTransaction(settingsA);
  digitalWrite(ncs, LOW);
}

void adns_com_end(){
  digitalWrite(ncs, HIGH);
  SPI.endTransaction();
}
--------------------------------------------------------------
After doing these changes, I got proper startup banner. 

However, I still don't get consistent position values. (Just 1, -1, 255 values)

What should be the clock rate to get proper values? 

Or what else should I try to make the sensor work? 

Thanks 
Alp

ScienceKyle, 2018-11-09:
What did you figure out? I am using this with a teensy 3.6 and your changes to the SPI allowed me to get the correct startup banner and what seems like reasonable data (corresponds to movement). It seems like the +- values relate to how much the sensor has moved since the last time step (i'm using the polling code).

aacemoglu, 2018-11-14:
I had a problem with attach interrupt function of Arduino. When I commented
it, it started to work as you were saying.
Just do not use this syntax ' attachInterrupt(digitalPinToInterrupt(pin),
ISR, mode); (recommended) ' which was recommended in this link
https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/
.

ScienceKyle <notifications@github.com>, 9 Kas 2018 Cum, 06:59 tarihinde
şunu yazdı: