mrjohnk/PMW3360DM-T2QU #22: Pointer doesn't respond to quick motion

raycewest, 2018-11-10:
Trying to use this for a trackball.  Config1 is set to 0x1, I have to keep it low to get results that remotely resemble mouse movement.  Otherwise, all of the other setup for the interrupt example is the same.  The side of the lens (flush) facing the ball is 2.4mm away, with about a 5x10mm hole exposed for sensing.  Are there any any registers/features I need to set up for good mouse-sensing?  Reading over the PMW/ADNS data sheets, nothing is jumping out at me.  Thanks for any advice!

mrjohnk, 2018-11-10:
You may have a problem with the ball.  Some surfaces track better than
others.  I had similar results with some materials including pool balls and
large steel ball bearings.  The surface needs to have trackable artifacts
and not be shiny.  I ended up with a Delrin ball that I bought from
McMaster-Car.  It has a sort of orange peel look, but is smooth.

raycewest, 2018-12-26:
Made a new holder and corrected the positioning of the sensor per the data sheet (so that the lens center, not the middle of the board, is centered directly above the ball), and it's dang close to exactly 2.4 mm away.  Also reduced the size of the view hole per the data sheet mins/maxes - https://imgur.com/a/r1AoouZ.  If anything looks wrong with it, or the ball, please let me know - otherwise I plan on purchasing a different ball since I still have this weird issue;  I'm at a complete loss as to what could be causing it.  Thanks/happy holidays!

mrjohnk, 2018-12-31:
> Also just tried using it a as regular mouse (with a surface instead of a
> ball) and reproduced the issue. So I doubt it's the ball. Could I shoot you
> my modifications to the code?

You can send it over and I'll take a look.  Also, what microprocessor board
are you using?  Email:John@MrJohnK.com

raycewest, 2018-12-31:
Thanks! Just sent it; however, I also tried from the base interrupt code and only modifying the config 1 value to 0x1 and uncommenting the mouse.move command.  Same result.  

I'm using a pro micro 5v.  Perusing the data sheet last night, I did notice the max VIN values for the PMW, which I see are lower than the ADNS.  Not sure how that could cause the issue though.

raycewest, 2019-01-07:
So, I'm starting to think it might be something to do with interrupt reporting - since seeing the voltages numbers, I pulled out the ADNS again and realized it has the same issue on the interrupt side - but polling seems to work fine.  I then tried polling on the PMW, and instead of the original issue, the pointer seems much too sensitive and erratic - different problem entirely.  I wonder if there could somehow be a timing issue with interrupts?  But I imagine others have used the interrupt code with no issues....

mrjohnk, 2019-01-07:
I've not seen any issues with the code.  The interrupt does depend on
having a pull-up resistor on the MOT line.  Maybe try disabling the pull-up
in the code and add your own 10K resistor pull-up?

SunjunKim, 2019-03-29:
I dug this issue and made a set of well-behaving examples with burst mode operation.
Please check my project:
https://github.com/SunjunKim/PMW3360_Arduino

raycewest, 2019-03-29:
I definitely will; thank you!